h2zero/n-able-Arduino #1: Collaborators wanted!

h2zero, 2021-11-27:
If anyone would like to help build this up for a release please comment in this issue thread. 

Todo: 

- add more boards
- add documentation 
- add wrapper library for FreeRTOS
- add github actions
- convert remaining nRF5 legacy code to use nrfx
- PlatformIO integration
- build platform_index.json

freylax, 2021-12-31:
I would like to contribute, need it for the meshtastic.org nrf52 integration. It is based on PlatformIO, so I would concentrate on this first. I would appreciate hints for doing this, I'm new to all this environments..

h2zero, 2021-12-31:
Great, thanks for the interest. I have started on the platform, you will find it here https://github.com/h2zero/platform-n-able

There is an example in there that you can use. This is very rough at the moment and support needs to be added to import the macro definitions from the boards.txt file,  especially the `__STACK_SIZE` and `__HEAP_SIZE`. I have hard coded it [here](https://github.com/h2zero/platform-n-able/blob/c9a0129b85673651c2b3ec4612ef7cce1bef0085/builder/frameworks/arduino/nrf5.py#L63) but that is only to test and needs to be removed/replaced.

We also need to create a framework script for nRF52 boards, idea's to do this can be borrowed from here https://github.com/platformio/platform-nordicnrf52.

It's a rough start but it's a start.

freylax, 2022-01-06:
So what I did find out it, that the settings of the boards.txt file of the framework repo like eg framework-arduinofruitnrf52 
have their corresponding settings in the boards/the_specific_bord.json files in the platform repo. In the case of the __STACK_SIZE it belongs to the extra_flags build settings. How this mapping from boards.txt to .json was done I do not know, if it is hand made or done by a script...  But for first we could try by hand withe the waveshare board.

h2zero, 2022-01-06:
Great! Thanks for digging into this 😀. That link is helpful for sure.

freylax, 2022-01-07:
There is no automatic conversion process from arduino to platformio, because
it is not 100% compatible, how to deal with the arduino menu settings...
https://github.com/stm32duino/Arduino_Core_STM32/pull/924#issuecomment-588193728

h2zero, 2022-01-08:
I suspected that, great investigative work! 

Looks like we need to define the base values for `__STACK_SIZE` and `__HEAP_SIZE` in the extra flags for the each board using the `extra_flags` in the json. This shouldn't be too much of a problem as the values would be the same for each chip model. I'll look into this a bit as well.

freylax, 2022-01-08:
I am happy when we manage to see things more clearly!
__STACK_SIZE and __HEAP_SIZE comes from your nimble implementation right? It is defined on the mynewt implementatioin as well I suppose? How is it done there, I will have a look.
I also will look how the current nrf5 platformio arduino handles the board variants which are menu entries in the boards.txt file, which could be easily detected by just looking at the according .json board files.

h2zero, 2022-01-08:
> __STACK_SIZE and __HEAP_SIZE comes from your nimble implementation right? It is defined on the mynewt implementatioin as well I suppose?

Those are linker macros actually, used here to set the sizes to maximize available ram, they don't have anything to do with NimBLE. When we define the on the command line the linker uses those values instead of the ones in the script. So all we need is to define them where they will be added to the command.