h2zero/n-able-Arduino #4: Platformio framework/platform not yet in registry.

h2zero, 2022-08-08:
I added this as a new framework to the Platformio registry (also the n-able platform) at the same time of the first release (3 days ago), not sure why it's not yet available (maybe @ivankravets, could comment)?

In the meantime to use this with Platformio you'll need to setup your environment as follows:

```
[env]
platform = https://github.com/h2zero/platform-n-able.git#1.0.0
framework = arduino
board = ...
...
```

ivankravets, 2022-08-09:
Hi @h2zero,

Thanks for using PlatformIO and for your willingness to make a contribution. 

You need to make PR to https://github.com/platformio/platform-nordicnrf52 . Otherwise, developers will see multiple duplicate boards. Dev-platform is based on MCU/Architecture, but not on a framework. We have multiple dev-platforms which support the Arduino framework.

See how do we handle multiple Arduino cores for nRF5 https://github.com/platformio/platform-nordicnrf52/tree/develop/builder/frameworks

h2zero, 2022-08-09:
@ivankravets thanks for the response. 

That makes sense, I'm not sure how much of the platform changes I made would be PR-able since the board configs were completely reworked.

Oh well, I'll think on it, thanks!

ivankravets, 2022-08-09:
Ahh, I see. You can instruct users to use 

```
[env]
platform = https://github.com/h2zero/platform-n-able.git#1.0.0
```

So, in this case, you don't need to publish a new dev-platform.

h2zero, 2022-08-09:
Yes, that's probably the way to go. I did make a few changes that might be of benefit to PR. 

Also, a bit off topic, but since you're here the `nrfjprog` tool on platformio is a bit old and doesn't support erasing the nrf52833. I worked around this by adding erase commands for all the tools instead but I thought you might want to update that one.

ivankravets, 2022-08-09:
>  but since you're here the nrfjprog tool on platformio is a bit old and doesn't support erasing the nrf52833. 

Could I ask you to file a feature request at https://github.com/platformio/platform-nordicnrf52 ?

h2zero, 2022-08-09:
> Could I ask you to file a feature request at https://github.com/platformio/platform-nordicnrf52 ?

Done: https://github.com/platformio/platform-nordicnrf52/issues/158

closed 2025-06-12 by h2zero