h2zero/n-able-Arduino #6: Adafruit Feather nRF52840 Express compilation error

Xraydylan, 2022-08-25:
I am having a problem with the Adafruit Feather nRF52840 Express board (it extends to all Adafruit nRF52840 boards).

When compiling any sketch, even an empty sketch, I get the many errors from the TinyUSB library in the `dcd_nrf5x.c` file.
As far as I could figure it out, constants of the nordic library, more specific the in files "nrf52840.h" and "nrf52840_bitfields.h", cant be found within `dcd_nrf5x.c`.

These are the error messages:

```
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c:50:32: error: 'USBD_INTEN_ENDEPIN0_Pos' undeclared here (not in a function)
   50 |   EDPT_END_ALL_MASK = (0xff << USBD_INTEN_ENDEPIN0_Pos) | (0xff << USBD_INTEN_ENDEPOUT0_Pos) |
      |                                ^~~~~~~~~~~~~~~~~~~~~~~
"C:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\tools\\gcc-arm-none-eabi\\9.3.1-1/bin/arm-none-eabi-gcc" -mcpu=cortex-m4 -mthumb -c -g -Os -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -MMD -DF_CPU=64000000 -DARDUINO=10819 -DARDUINO_NRF52840_FEATHER -DNRF52_SERIES -DNRF52832_XXAA -DUSE_ADA_BL -DUSBCON -DUSE_TINYUSB -DUSB_VID=0x239A -DUSB_PID=0x8029 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather nRF52840 Express\"" -DCONFIG_NIMBLE_CPP_LOG_LEVEL=0 -DNRF5 -DARDUINO_ARCH_NRF5 "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5/nimble_config" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5/nordic/nrfx" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5/nordic/nrfx/hal" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5/nordic/nrfx/mdk" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5/nordic/nrfx/soc" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5/nordic/nrfx/drivers/include" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5/nordic/nrfx/drivers/src" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5/CMSIS/Include" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5/FC_Store" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5/BLEBond_nvs" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5/TinyUSB" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src" -DUSE_LFXO "@C:\\Users\\Dylan\\AppData\\Local\\Temp\\arduino_build_521375/build.opt" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5" "-IC:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\variants\\feather_nrf52840_express" "C:\\Users\\Dylan\\AppData\\Local\\Arduino15\\packages\\h2zero\\hardware\\arm-ble\\0.1.0\\cores\\nRF5\\nordic\\nrfx\\mdk\\system_nrf52840.c" -o "C:\\Users\\Dylan\\AppData\\Local\\Temp\\arduino_build_521375\\core\\nordic\\nrfx\\mdk\\system_nrf52840.c.o"C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c:50:68: error: 'USBD_INTEN_ENDEPOUT0_Pos' undeclared here (not in a function)

   50 |   EDPT_END_ALL_MASK = (0xff << USBD_INTEN_ENDEPIN0_Pos) | (0xff << USBD_INTEN_ENDEPOUT0_Pos) |
      |                                                                    ^~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c:51:23: error: 'USBD_INTENCLR_ENDISOIN_Msk' undeclared here (not in a function); did you mean 'CCM_INTENCLR_ENDKSGEN_Msk'?
   51 |                       USBD_INTENCLR_ENDISOIN_Msk | USBD_INTEN_ENDISOOUT_Msk
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                       CCM_INTENCLR_ENDKSGEN_Msk
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c:51:52: error: 'USBD_INTEN_ENDISOOUT_Msk' undeclared here (not in a function)
   51 |                       USBD_INTENCLR_ENDISOIN_Msk | USBD_INTEN_ENDISOOUT_Msk
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c: In function 'edpt_dma_start':
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c:119:50: error: 'USBD_IRQn' undeclared (first use in this function); did you mean 'ECB_IRQn'?
  119 |       if ( __get_PRIMASK() || !NVIC_GetEnableIRQ(USBD_IRQn) )
      |                                                  ^~~~~~~~~
      |                                                  ECB_IRQn
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c:119:50: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c:127:19: error: 'NRF_USBD' undeclared (first use in this function)
  127 |           ended = NRF_USBD->EVENTS_ENDISOIN + NRF_USBD->EVENTS_ENDISOOUT;
      |                   ^~~~~~~~
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c: In function 'xact_out_prepare':
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c:170:5: error: 'NRF_USBD' undeclared (first use in this function)
  170 |     NRF_USBD->TASKS_EP0RCVOUT = 1;
      |     ^~~~~~~~
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c: In function 'xact_out_dma':
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c:187:28: error: 'NRF_USBD' undeclared (first use in this function)
  187 |   uint8_t const xact_len = NRF_USBD->SIZE.EPOUT[epnum];
      |                            ^~~~~~~~
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c: In function 'xact_in_prepare':
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c:210:3: error: 'NRF_USBD' undeclared (first use in this function)
  210 |   NRF_USBD->EPIN[epnum].PTR    = (uint32_t) xfer->buffer;
      |   ^~~~~~~~
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c: In function 'dcd_int_enable':
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c:229:18: error: 'USBD_IRQn' undeclared (first use in this function); did you mean 'ECB_IRQn'?
  229 |   NVIC_EnableIRQ(USBD_IRQn);
      |                  ^~~~~~~~~
[... cut ...]
      |                                                                                            ^~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Dylan\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.1.0\cores\nRF5\TinyUSB\Adafruit_TinyUSB_ArduinoCore\tinyusb\src\portable\nordic\nrf5x\dcd_nrf5x.c:821:28: error: 'USBD_IRQn' undeclared (first use in this function); did you mean 'ECB_IRQn'?
  821 |       NVIC_ClearPendingIRQ(USBD_IRQn);
      |                            ^~~~~~~~~
      |                            ECB_IRQn
exit status 1
Error compiling for board Adafruit Feather nRF52840 Express.

```

h2zero, 2022-08-25:
Thanks for the report! I have not encountered this issue before, I wonder if there is a conflict caused by something in your environment? When first building do you see any messages about the libraries used?

Xraydylan, 2022-08-26:
I have removed all libraries from Arduino (except for Nimble) but the errors still appear.

h2zero, 2022-08-26:
Thanks! I have confirmed the problem, it seems I had updated the boards.txt file locally and it wasn't pushed with the release. I will fix this shortly and create a new release.

h2zero, 2022-08-26:
Sorry for the trouble! I've just created a new release to resolve this and is available to update from board manager.

Xraydylan, 2022-08-26:
Wonderful!
I will test it later.

Xraydylan, 2022-08-26:
Unfortunately it didn't help much...

I am now getting this error:
```
exec: "adafruit-nrfutil": executable file not found in %PATH%
Error compiling for board Adafruit Feather nRF52840 Express.
```

h2zero, 2022-08-26:
You'll need to install adafruit-nrfutil, I did not include it in the boards package due to size and other issues. 

You'll need to install it with`pip3 install --user adafruit-nrfutil` or the other methods described here: https://github.com/adafruit/Adafruit_nRF52_nrfutil

h2zero, 2022-08-26:
I should have also mentioned that if you don't plan to use the bootloader, then you can skip this and just select `none` in the tools menu -> bootloader. You'll need to use Jlink or one of the other debuggers to upload of course.

Xraydylan, 2022-08-26:
It is perfectly working now. 
Thanks for your help!

h2zero, 2022-08-26:
Awesome! thanks again for the report and you're very welcome!