h2zero/n-able-Arduino #14: Error. Board type requires board_bootloader to be specified

asukiaaa, 2022-12-22:
I tried to build for feather ntf52832 with the following configuration but cannot build.

```
[env:n-able]
platform = https://github.com/h2zero/platform-n-able.git
framework = arduino
board = adafruit_feather_nrf52832
```

Logs
```
Processing n-able (platform: https://github.com/h2zero/platform-n-able.git; framework: arduino; board: adafruit_feather_nrf52832)
---------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/n-able/adafruit_feather_nrf52832.html
PLATFORM: n-able (1.0.0+sha.79f9fef) > Adafruit Bluefruit nRF52832 Feather
HARDWARE: NRF52832_XXAA 64MHz, 64KB RAM, 496KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-n-able-arduino @ 0.1.0+sha.a6263a7 
 - tool-adafruit-nrfutil @ 1.503.0 (5.3) 
 - tool-sreccat @ 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi @ 1.90301.200702 (9.3.1)
Error. Board type requires board_bootloader to be specified
====================================== [FAILED] Took 0.73 seconds
```

Is there any idea to resolve the error?

asukiaaa, 2022-12-25:
I could run building process by specifying `board_bootloader`.
I found the setting on example of [platform-n-able](https://github.com/h2zero/platform-n-able).
https://github.com/h2zero/platform-n-able/blob/fee357702d138dfc7f1b023aefdc5b6cc36e57eb/examples/BLE-client-test/platformio.ini#L60-L64

```ini
[env:n-able]
platform = https://github.com/h2zero/platform-n-able.git
framework = arduino
board = adafruit_feather_nrf52832
board_bootloader = adafruit
```

I could build and write my program but it does not work.
(Does not blink LED_BUILTIN and does not send any  info over USB serial.)

Do I need to write bootloader with using STLink or some for n-able?

There is no issue thread on https://github.com/h2zero/platform-n-able so I continue discussion here.

h2zero, 2022-12-27:
if you are not using a bootloader you should specify `board_bootloader = adafruit`

asukiaaa, 2022-12-29:
I tried `none` for `board_bootloader` but still cannot control LED and Serial.

h2zero, 2022-12-30:
I understand now, do you still have the original bootloader on the board? You will need to flash with stlink/jlink or some other debugger so that your sketch will be loaded and correctly. You can flash with no bootloader as an option or use the one provided here to enable flashing over serial without a debugger attached.

asukiaaa, 2022-12-30:
Thank you for the information.
I understood that I need some writer(debugger) to upload boot loader for n-able.