h2zero/n-able-Arduino #16: OTA DFU -> any Plans for Future?

ericlangel, 2023-02-14:
is there a working OTA DFU Example?

h2zero, 2023-02-20:
This is not yet implemented as it requires a different bootloader (new), I hope to have this sometime in the future.

intakenick, 2024-01-24:
Hey @h2zero, what are your current thoughts on this? Is it possible to use the NRF5 bootloader from Nordic in conjunction with n-able core? I imagine there wouldn;t be room for both the softdevice and nimble stack.

h2zero, 2024-01-25:
The two can't be used together, I had thoughts of making a mcuboot based bootloader but have t had time to work on it.

intakenick, 2024-01-25:
Thanks. To your knowledge, is it possible to ota without a bootloader?

h2zero, 2024-01-27:
Well I suppose you may be able to construct something if the flash is properly partitioned. You would need to know what partition to jump to somehow, usually a record in flash somewhere that indicates which partition is valid. Would be a bit of work and would require modifications to the linker files.