h2zero/n-able-Arduino #17: nrf52810 - no fpu

bitmat1, 2023-02-28:
I am unable to get this to compile for an nrf52810 as it has no floating point unit.
Is it possible to change the freertos to compile without FPU??

I've added an entry for it in Boards.txt and added system_nrf52810.h and .c files to the nrf mdk, but when i select  "generic 52810" I get this error:

#error This port can only be used when the project options are configured to enable hardware floating point support

Weirdly, if I select generic nrf51822 as the board then it compiles and runs ok on this nrf52810 custom board, and example code shows that a bluetooth device is available. Should I just use this, or will all the pins and interrupts be wrong?

This post gives a solution but asm format is different and mighty confusing! https://devzone.nordicsemi.com/f/nordic-q-a/30103/freertos-on-nrf52810

h2zero, 2023-03-01:
> There was an "undefined reference to `SystemInit'" in gcc_startup_nrf52810.S:308: but after removing bl SystemInit on line 308 it all works. The function systemInit is defined in system_nrf52810, is it needed for this chip?

That file will need to be added so that the startup is correct. 

> Should I make a nrf52810_nimconfig.h file next?

That would be advised for sure. 

If you don't mind sharing, I would like to ask that you submit a PR with your changes so others may benefit as well.

bitmat1, 2023-03-01:
Yes absolutely, I've not made a pull request before, but i will do so and test it once i have this working. 
I've made the nrf52810_nimconfig.h based on the nrf52832 one and now its giving me these errors when compiling the nimBLE client example: nrf_nvm_store.c:19: undefined reference to `nrfx_nvmc_page_erase' and nrf_nvm_store.c:40: undefined reference to `nrfx_nvmc_words_write'.  But it seems like i'm very close to getting it working now.
Sorry for my ignorance, I am an electronics engineer who dabbles in coding and am way over my head with this.
How and where should I include system_nrf52810.h and nrfx_nvmc.h so these references are defined?

bitmat1, 2023-03-01:
I figured it out! I had copied the system_nrf52810 file from another copy of the mdk and wasn't correct and had to make an edit to the nrfx_config_nrf52810 file to enable nvmc. Now it's working and I'm very happy that i worked this out with your help. I will make a pull request. Thanks.

h2zero, 2024-07-05:
@bitmat1 Support now officially added and hopefully fully functional via #46 and #54.

closed 2024-07-05 by h2zero