h2zero/n-able-Arduino #22: Need help with deep sleep NRF51822 board

MegaSegaDrive8, 2023-06-25:
Hello. Thank you for this library.

I need help with deep sleep with board nrf51822. 
Now for example "delay(5000);" get me about 0.8-0.9 mA. 

Is this maximum mA for sleep mode?

h2zero, 2023-06-29:
You're welcome 😄 , is there anything else on the board drawing current? On my nrf51822 boards the current draw is much less in sleep mode. What peripherals are you using?

MegaSegaDrive8, 2023-06-29:
No, just 2 contacts like switch, between GND and P16

h2zero, 2023-06-29:
I don't have any examples of that at the moment but I am wondering if you are calling `Serial.begin()` in your code? the UART will consume power at that level.

h2zero, 2023-11-16:
@beckmx `systemPowerOff()` will put the device into deep sleep while just calling `delay()` will put the device into "system on" sleep mode.