h2zero/n-able-Arduino #32: Crash when loop is under 10ms on NRF51802 (xaa)

doh38, 2023-11-18:
I have a device with a nrf51802. 
just calling anything under 10ms (even with the simplest form with delay(9) in the main loop) will make a few iteration and crash. funnily calling a delayMicrosecond(100000) will also crash it. 
I don't intent to use "delays" actually, but even the freertos example crash or any millis() usage. 
Is there anything I can do to debug this? 
I don't have serial on this board and use RTT... unfortunately here on an higher level, so I'm not sure to be able to check anything there :)

doh38, 2023-11-19:
by default the watchdog is activated. first time ever used. 
You need to disable it or add `NRF_WDT->RR[0] = WDT_RR_RR_Reload; at any place you want it to be reseted. 
so solved :)

jhmaloney, 2023-12-03:
I also ran into this issue. You can also disable the watchdog timer by adding:
`-DCONFIG_WDT_TIMEOUT_SECONDS=0`
to your compiler flags.

It would be nicer for newcomers to n-able-Arduino if the CONFIG_WDT_TIMEOUT_SECONDS setting defaulted to 0 to disable the watchdog timer.

Alternatively, 'NRF_WDT->RR[0] = WDT_RR_RR_Reload' could be added to the top-level loop in main.c.

jhmaloney, 2024-02-24:
I just submitted a tiny PR to make CONFIG_WDT_TIMEOUT_SECONDS default to zero (WDT disabled).