h2zero/n-able-Arduino #35: Custom NRF51822 QFAA will advertise but won't connect

intakenick, 2023-12-11:
Hi there @h2zero - I'm having an issue getting my custom design to connect using NRFconnect. It's odd: all of the other functionality works (GPIO, sleep, i2c) and my BLE server test code works as far as it shows up in advertising but fails to connect. I've tried the same code on a cheap generic development board and it works fine. The schematic is identical minus the 32khz crystal (I'm not using one in my design). Any idea what might be happening here? I'm using a 10ppm 16mhz crystal for the main clock.

h2zero, 2023-12-11:
Not sure what is happening there, what are you using for the LF clock source? Do you have another board you can try this on?

intakenick, 2024-01-04:
I discovered the cause of this issue: the capacitors on my crystal were chosen based on Nordic's nrf51822 reference design (22pF) and not to the datasheet (12pF). Swapping the XTAL caps did the trick. Thanks for your great work on this core!

closed 2024-01-04 by intakenick

h2zero, 2024-01-09:
Awesome, glad you found the issue!