h2zero/n-able-Arduino #45: nRF52810 support?

positron96, 2024-06-17:
Hello. Does this core support nRF52810 MCU? I am doing a custom board with that MCU, and looking for PlatformIO framework to use.

I don't see it mentioned in the list ofsupported boards, but I understand that the compiler, tools and libraries are the same as for other nRF52 MCUs. How extensive will the changes needed to compile for the that MCU be?

h2zero, 2024-06-17:
It's no too extensive, would need to pull in the startup code from nrfx and adjust the linker, and a board variant.

positron96, 2024-06-18:
Would you be interested in a PR if I make it work for me?

h2zero, 2024-06-18:
I would love it if you did :). If you want to start I can help as well. Just make a change and open a PR that we can both add to.

PR #46 (2024-06-23): Add support for nRF52810
Fixes #45 
Uses info from #17 
Also needs patch to PlatformIO python script (it adds FPU compiler flags), see https://github.com/h2zero/platform-n-able/pull/1

I was able to compile blink and basic BLE example (BLE-server-test).

Here are some comparisons with other cores (neither work without modifications):

**blink:**
n-able:
```
RAM:   [==        ]  20.3% (used 4988 bytes from 24576 bytes)
Flash: [=         ]   9.1% (used 17896 bytes from 196608 bytes)
```

zephyr (probably contains BLE stack, as numbers are not much different from below):
```
RAM:   [======    ]   [... cut]
closed 2024-06-23 by h2zero