h2zero/n-able-Arduino #51: NRF52810 BLE Stack Overflow When Writing to Characteristic

jackb60, 2024-07-04:
I am testing the BLE_uart.ino example with the NRF52810. The Tx characteristic works fine, but writing to the Rx characteristic causes a stack overflow error, interestingly after the data is received. Here is the full debug output:

```
DD NimBLECharacteristic: Characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e Write event
D NimBLECharacteristic: >> setValue: length=1, data=41, characteristic UUID=6e400002-b5a3-f393-e0a9-e50e24dcca9e
D NimBLECharacteristic: << setValue
*********
Received Value: A
*********
D NimBLECharacteristicCallbacks: onWrite: default
Task: ble, stack overflow
```

Sometimes I get the exception number as well, for example:
```
D NimBLECharaD NimBLECharacteristic: Characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e Write event
D NimBLECharacteristic: >> setValue: length=1, data=ff, characteristic UUID=6e400002-b5a3-f393-e0a9-e50e24dcca9e
D NimBLECharacteristic: << setValue
*********
Received Value: �
*********
D NimBLECharacteristicCallbacks: onWrite: default
TaskUnhandled exception 0x00000003 , exception sp 0x20005e50
R0: 0x00000001, , R1: 0x00000000 , R2: 0x20001dfc, R3: 0x0000f900, R12: 0x00000080
LR: 0x00017509, PC: 0x0001750e, PSR: 0x01000011
```

jackb60, 2024-07-04:
changing line 67 of nrf52810_nimconfig.h from
 `#define CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE 1408` to 
`#define CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE 2048` fixed this issue.

h2zero, 2024-07-05:
This core was designed for these settings to be adjustable, you don't need to change the values in the config files directly. Open Arduino and open your sketch, add a file called `build_opt.h` and copy/paste this in it:
```
-DCONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE=2048
```
That will override the value in the nimconfig file. If you want to change any other value just add another line and put: 
```
-DWHATEVER_CONFIG_DEFINE_HERE=?
```

closed 2024-07-05 by jackb60