h2zero/n-able-Arduino #52: NRF52810 SPI/I2C

jackb60, 2024-07-04:
As is, neither the SPI or Wire library compile for the NRF52810. 
Error for SPI:
```
C:\Users\jackb\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.2.0\libraries\SPI\SPI.cpp:258:15: error: 'NRF_SPI2' was not declared in this scope; did you mean 'NRF_SWI2'?
  258 | SPIClass SPI (NRF_SPI2,  PIN_SPI_MISO,  PIN_SPI_SCK,  PIN_SPI_MOSI);
      |               ^~~~~~~~
      |               NRF_SWI2

exit status 1

Compilation error: exit status 1
```
Error for I2C:
```
C:\Users\jackb\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.2.0\libraries\Wire\Wire_nRF52.cpp:402:14: error: 'NRF_TWIM1' was not declared in this scope; did you mean 'NRF_TWIM0'?
  402 | TwoWire Wire(NRF_TWIM1, NRF_TWIS1, SPIM1_SPIS1_TWIM1_TWIS1_SPI1_TWI1_IRQn, PIN_WIRE_SDA, PIN_WIRE_SCL);
      |              ^~~~~~~~~
      |              NRF_TWIM0
C:\Users\jackb\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.2.0\libraries\Wire\Wire_nRF52.cpp:402:25: error: 'NRF_TWIS1' was not declared in this scope; did you mean 'NRF_TWIS0'?
  402 | TwoWire Wire(NRF_TWIM1, NRF_TWIS1, SPIM1_SPIS1_TWIM1_TWIS1_SPI1_TWI1_IRQn, PIN_WIRE_SDA, PIN_WIRE_SCL);
      |                         ^~~~~~~~~
      |                         NRF_TWIS0
C:\Users\jackb\AppData\Local\Arduino15\packages\h2zero\hardware\arm-ble\0.2.0\libraries\Wire\Wire_nRF52.cpp:402:36: error: 'SPIM1_SPIS1_TWIM1_TWIS1_SPI1_TWI1_IRQn' was not declared in this scope
  402 | TwoWire Wire(NRF_TWIM1, NRF_TWIS1, SPIM1_SPIS1_TWIM1_TWIS1_SPI1_TWI1_IRQn, PIN_WIRE_SDA, PIN_WIRE_SCL);
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```
I was able to get the I2C library working by changing line 402 in Wire_nRF52.cpp from 
`TwoWire Wire(NRF_TWIM1, NRF_TWIS1, SPIM1_SPIS1_TWIM1_TWIS1_SPI1_TWI1_IRQn, PIN_WIRE_SDA, PIN_WIRE_SCL);` to 
`TwoWire Wire(NRF_TWIM0, NRF_TWIS0, TWIM0_TWIS0_IRQn, PIN_WIRE_SDA, PIN_WIRE_SCL);`

Obviously this would not be compatible with other NRF52 modules, so I think some sort of compiler flag needs to be used but I don't have the programming skills for that yet.

I have been unable to get the SPI library to work. I changed line 258 in SPI.cpp from 
`SPIClass SPI (NRF_SPI2,  PIN_SPI_MISO,  PIN_SPI_SCK,  PIN_SPI_MOSI);` to
`SPIClass SPI (NRF_SPI0,  PIN_SPI_MISO,  PIN_SPI_SCK,  PIN_SPI_MOSI);`

and that compiles, but SPI.transfer() hangs.

h2zero, 2024-07-04:
Thanks! I wasn't able to test this as I don't have a device so the feedback is very valuable. I will add your fixes to a PR for you to review.

h2zero, 2024-07-05:
Are we able to close this now @jackb60 ?

closed 2024-07-05 by jackb60