h2zero/n-able-Arduino #61: No output from nrf51822's Serial

goddade, 2024-12-23:
The RX of the serial debugger is connected to P0.22, and the TX is connected to P0.21.

```
#include <Arduino.h>

#define PIN_RX (22)
#define PIN_TX (23)

void setup(void) {
    Serial.setPins(PIN_RX, PIN_TX);
    Serial.begin(115200);
    Serial.printf("setup\n");
}

void loop() {
    /** Loop here and send notifications to connected peers */
    Serial.println("loop");
    delay(200);
}
```

goddade, 2024-12-28:
> Hello @goddade I am not able to reproduce this issue, do you perhaps have a hardware problem?

I connected the wrong pins.

I don't quite understand g_ADigitalPinMap, why should P0.22 be written as 23?

h2zero, 2025-01-02:
Not sure I follow, which board are you using? Most nRF51 boards have 1:1 pin mapping, so 22 = 22 etc

goddade, 2025-01-12:
When I set `board = nrf51_beacon`, 
`g_ADigitalPinMap` located in `.platformio\packages\framework-n-able-arduino\variants\PCA20006\variant.cpp` is as follows:
```
const uint32_t g_ADigitalPinMap[] = {
  0,
  1,
  2,
  3,
  4,
  5,
  5,
  6,
  7,
  8,
  9,
  10,
  11,
  12,
  13,
  14,
  15,
  16,
  17,
  18,
  19,
  20,
  21,
  22,
  23,
  24,
  25,
  (uint32_t)-1,
  (uint32_t)-1,
  28,
  29,
  30,
  (uint32_t)-1
};
```
There are two consecutive 5. Is this a bug?

h2zero, 2025-01-12:
Oh, thank you! Yes that is certainly a bug.

closed 2025-01-13 by goddade