h2zero/n-able-Arduino #63: How to debug Unhandled exception (HardFault)?

tsl0922, 2025-01-04:
Hi, I'm running the `NimBLE_Server` example on my nrf51822-qfaa board, but when I send some data to the server, it always trigger a restart (HardFault), How do I know where is the code that causes the exception?

Serial log:

```
Starting NimBLE Server
Advertising Started
Client address: 45:a2:be:5a:39:b7
MTU updated: 23 for connection ID: 1
0xf00d : onRead(), value: Fries
0xf00d : onWrite(), value: abcd
Unhandled exception 0x00000003 , exception sp 0x20001510
R0: 0x00000000, , R1: 0x2000098c , R2: 0x00000001, R3: 0xffffffff, R12: 0x00000066
LR: 0x00009111, PC: 0x00003946, PSR: 0x61000000
Starting NimBLE Server
Advertising Started
```

h2zero, 2025-01-04:
You can use GDB with a debugger probe if you have one. Otherwise GCC address2line https://linux.die.net/man/1/arm-linux-gnu-addr2line. Or just look at your map file and find where the sp is pointing.

tsl0922, 2025-01-05:
Got a line number with `addr2line`, the exception happens in nimble code.

```
Starting NimBLE Server
Advertising Started
Client address: b8:e8:56:06:23:4b
MTU updated: 23 for connection ID: 1
62750002-d828-918d-fb46-b6c11c675aec : onWrite(), value: ␁
Unhandled exception 0x00000003 , exception sp 0x200019e0
R0: 0x2000151c, , R1: 0x2000151c , R2: 0x20001de8, R3: 0x000000c1, R12: 0x00000004
LR: 0x0000ee5d, PC: 0x0000ee44, PSR: 0x21000000
```

```
arm-none-eabi-addr2line -e firmware.elf -a 0x0000ee44
0x0000ee44: os_mbuf_free at .../.pio\libdeps\epd420\NimBLE-Arduino\src\nimble\porting\nimble\src/os_mbuf.c:321
```

tsl0922, 2025-01-05:
Maybe memory issue.

Replacing the n51822_xxaa board with a n51822_xxac one works fine.

tsl0922, 2025-01-05:
Changing `CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE` to `2048` fixes the issue on n51822_xxaa.

You may make `2048` to the default on nrf51 too,  @h2zero.

closed 2025-01-05 by tsl0922