h2zero/n-able-Arduino #66: The program stops at 0x0000000000000 on NRF51822 CEAAD2 chip

goddade, 2025-01-13:
main.cpp:
```
#include <Arduino.h>

// put function declarations here:
int myFunction(int, int);

void setup() {
  // put your setup code here, to run once:
  int result = myFunction(2, 3);
}

void loop() {
  // put your main code here, to run repeatedly:
}

// put function definitions here:
int myFunction(int x, int y) {
  return x + y;
}
```
platformio.ini:
```
[env:generic_nrf51822_xxaa]
platform = n-able
board = generic_nrf51822_xxaa
framework = arduino

debug_tool = blackmagic
upload_protocol = blackmagic
debug_port = COM7
upload_port = COM7
monitor_port = COM6

lib_deps =
    h2zero/NimBLE-Arduino
```
Is it because this chip doesn't have a bootloader?
I tried `bootloader = none` , but it didn't work.

[image]

goddade, 2025-01-13:
Thank you. 
It works after erasing flash.

goddade, 2025-01-13:
Off-topic question: When using the NimBle-Arduino library, is there a way to reduce flash usage? The nRF51822xxAB only has 128k flash.

h2zero, 2025-01-13:
You can change config settings to reduce it somewhat. Disabling roles that you do not use for instance. You'll find all the options here https://h2zero.github.io/NimBLE-Arduino/md__command__line__config.html