h2zero/n-able-Arduino #72: Seeed Studio XIAO nRF52840 error

Jasionf, 2025-05-19:
Why do I get this error when I compile the XIAO nRF52840, I don't find it in the documentation, is there a missing link somewhere?

[image]

h2zero, 2025-05-19:
Something is wrong with your build environment.

Jasionf, 2025-05-20:
I followed the docs, Clone this repo: git clone https://github.com/h2zero/n-able-Arduino.git, download the NimBLE libraries, open the example and compile, and I get this error, which step did I miss?

Jasionf, 2025-05-26:
hello h2zero ， i use arduino IDE

h2zero, 2025-05-26:
Thank you @Jasionf, it looks to me like you have selected the board from another arduino core and not this one. If you use platformio instead it will be easier to make sure you are using the correct Arduino core.

h2zero, 2025-06-12:
@Jasionf I've made a few updates to this variant and will prepare a release shortly, please try when that is complete.