h2zero/n-able-Arduino #73: `nrfx_clock_hfclk_start()` doesn’t keep HFCLK running — turns off after 4 seconds

timnicolas, 2025-05-27:
Hi,
I’m experiencing an issue where calling nrfx_clock_hfclk_start() successfully starts the HFCLK, but it turns off automatically after approximately 4 seconds, even when it should remain running.

Here is a minimal reproducible example — it blinks a red LED every time the HFCLK stops and needs to be restarted:
```cpp
#include <Arduino.h>
#include <nrfx_clock.h>

#define LED_R 22

void setup(void) {
  pinMode(LED_R, OUTPUT);
  digitalWrite(LED_R, LOW);
  nrfx_clock_hfclk_start();
  while (!nrfx_clock_hfclk_is_running());
}

void loop(void) {
  if (!nrfx_clock_hfclk_is_running()) {
    digitalWrite(LED_R, HIGH);
    delay(100);
    digitalWrite(LED_R, LOW);

    nrfx_clock_hfclk_start();
    while (!nrfx_clock_hfclk_is_running());
  }
  delay(10);
}
```
**Expected behavior:**
Once started, the HFCLK should stay running unless explicitly stopped.

**Observed behavior:**
The HFCLK stops automatically after ~4 seconds unless it’s manually restarted, even though nrfx_clock_hfclk_is_running() initially returns true.

**Additional info:**
	•	Platform: nRF52832
	•	Environment: n-able-Arduino
	•	Disabling configUSE_TICKLESS_IDLE had no effect

Is there a recommended way to force the HFCLK to stay on permanently within the framework, or to properly register it as a required resource?

h2zero, 2025-05-27:
You should use these functions instead: https://github.com/h2zero/n-able-Arduino/blob/3e493d8cbd55916c0fb08b7d38c4811313567559/cores/nRF5/wiring.c#L150C5-L150C26

`hw_clock_hfxo_request`
`hw_clock_hfxo_release`

These are wrappers that track the request count to keep it running.

timnicolas, 2025-05-28:
Thanks for the clarification! I wasn’t aware of that function.
I tested it, and it’s working perfectly now.
Really appreciate your help!