h2zero/n-able-Arduino #79: Bonding on nRF52840 SuperMini/Pro-Micro/Nice!Nano v2

hal2001gv, 2025-08-07:
Thanks for the great work. I have extended your system for the use of the boards above. Everything is working except one thing:

I use bonding for a secure connection:
```
NimBLEDevice::setSecurityAuth(true, true, true); 
NimBLEDevice::setSecurityInitKey(BLE_SM_PAIR_KEY_DIST_ENC |  BLE_SM_PAIR_KEY_DIST_ID);
NimBLEDevice::setSecurityRespKey(BLE_SM_PAIR_KEY_DIST_ENC |  BLE_SM_PAIR_KEY_DIST_ID);  
NimBLEDevice::setSecurityPasskey(123456);
NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY); // Display only passkey
```

Unfortunately, the bonded address is stored correctly, i.e. `NimBLEDevice::isBonded(...)` return `false` as well as `NimBLEDevice::getBondedAddress(...)` shows the wrong address. `NimBLEDevice::getNumBonds()` shows the correct number after bonding, i.e. something should be stored.

The same code compiled/flashed on an ESP32 works without issues.

Any advice? Thanks in advance.

h2zero, 2025-09-16:
Thanks, I see it now. Looks like the esp32 does triggers the re-pairing on connection before the onConnect callback, whereas the Nimble controller does not so the address in onConnect is the random address and not the ID address that you see in the bind list.

hal2001gv, 2025-09-17:
Thank you for the update. Can I do anything to get the correct ID address from ```connInfo``` in ```onConnect(...)```?

h2zero, 2025-09-18:
Yes, that is perfect.

h2zero, 2025-10-01:
You will probably need to increase the max bonds and cccds. https://github.com/h2zero/NimBLE-Arduino/blob/dfb4ac561a06797081be9e752902a6582e7f029e/src/nimconfig.h#L127

hal2001gv, 2025-10-02:
Thank you for the prompt advice. By increasing the ```CONFIG_BT_NIMBLE_MAX_CCCDS``` to ```9``` I am now able to bond three devices. :-)

closed 2025-10-02 by hal2001gv